GW-Robotics/2017-Firefighter #13: Write code to actuate the extinguisher

omn0mn0m, 2017-01-22:


Adrian-Haber, 2017-02-19:
@jwalrusisawesome do we have a design for the extinguisher so that we can write some psudocode?

omn0mn0m, 2017-02-19:
@Adrian-Haber In one of the closed issue Jeremey wrote some pseudo code.

omn0mn0m, 2017-03-21:
Pseudocode for this:
```
For values between 90 and 180:
     If flame detected:
          Set foundFlame to true
          Break out of for loop
     Move server up by for-loop value

If no flame found:
     Move servo to 90
     For values between 90 and 0:
     If flame detected:
          Set foundFlame to true
          Break out of for loop
     Move servo down by for-loop value
Else:
     Actuate the pump
     While foundFlame is true: do nothing
     Disable pump
```

omn0mn0m, 2017-04-20:
Written in Arduino, needs Python implementation