GW-Robotics/2017-Firefighter #27: Write code to stop in front of a wall and decide a direction to move

omn0mn0m, 2017-02-15:
Pseudocode:

if wall:
   check left and right ultrasonics
   move in direction of clearest path

Bwall72, 2017-02-19:
#Check left and right sonar, turn in that direction

#reading on sonar for when to stop moving
stopReading = none

#assign values later using tuple
leftPin = none
rightPin = none
centerPin = none()

#Declare two sonar sensors
center = Sonar(centerPin[0],centerPin[1],centerPin[2])

left = Sonar(leftPin[0],leftPin[1],leftPin[2])

right = Sonar(rightPin[0],rightPin[1],rightPin[2])

if center.get_distnace > stopReading:
    if right.get_distance > left.get_distance:
        #move/turn right
    else:
        #move/turn left

omn0mn0m, 2017-03-05:
Code has been written to follow a maze and make its way out using what is called the [left-handed wall following algorithm](https://en.wikipedia.org/wiki/Maze_solving_algorithm#Wall_follower). This needs to be changed from using limit switches to using the ultrasonic's reading_in_range() function. 

When the robot is within a certain range (probably 0 to 5 cm), then it is equivalent to a limit switch being triggered.

From SEAS Innovation Challenge R&D Showcase code:

```python
def navigate_maze(self):
        """Simple Left-Handed Wall Follower Algorithm"""

        if self.front_switch.get():
            arcade_drive(0.0, 0.0, -1.0)

        if self.left_switch.get():
            arcade_drive(0.0, 0.0, 1.0)

        if self.right_switch.get():
            arcade_drive(0.0, 1.0, 0.0)
            sleep(1.0)
            arcade_drive(1.0, 0.0, 0.0)
```

omn0mn0m, 2017-03-05:
For now turn by time until we have working gyro code.